UCSD-E4E/smartfin-fw3 #23: Recorder doesnt get correct file length from deployment

emilybthorpe, 2023-08-10:
**Describe the bug**
Recorder.cpp attempts to get the length of the current file from the current Deployment using `getLength()`. This function uses `fstat `to get the number of bytes in the current file. `getLength()` in Deployment gets the length correctly, and prints out the correct length to the terminal. Recorder.cpp does not get it correctly, instead just getting 0. When moving the `fstat` code to Recorder, it gets the number of bytes correction

**Expected behavior**
Recorder correctly gets the number of bytes from deployment

ntlhui, 2023-08-14:
@emilybthorpe has this been resolved in c882ca2?

emilybthorpe, 2023-08-14:
Yep! I'll close the issue.

closed 2023-08-14 by emilybthorpe