UCSD-E4E/smartfin-fw3 #90: docs: Data Upload

ntlhui, 2024-10-01:
- [ ] src/cellular/dataUpload.hpp:49: error: Member exitState(void) (function) of class DataUpload is not documented.
- [ ] src/cellular/dataUpload.hpp:50: error: Member can_upload(void) (function) of class DataUpload is not documented.
- [ ] src/cellular/dataUpload.hpp:47: error: Member initSuccess (variable) of class DataUpload is not documented.
- [ ] src/cellular/dataUpload.hpp:48: error: Member lastConnectTime (variable) of class DataUpload is not documented.
- [ ] <can_upload>:1: error: parameters of member DataUpload::can_upload are not documented
- [ ] <exitState>:1: error: parameters of member DataUpload::exitState are not documented

ed2263e1fbb0b66328a2df6ef0a8531383ddd825

Tylody, 2024-11-15:
Seems like DataUpload lastConnectTime (type of system_tick_t) has no uses outside of dataUpload.hpp. Should I remove it?

ntlhui, 2024-11-15:
> Seems like DataUpload lastConnectTime (type of system_tick_t) has no uses outside of dataUpload.hpp. Should I remove it?

`lastConnectTime` is likely a remnant or incomplete implementation of rate limiting calls to `Particle.connect` - previously, we could not do more than 1 call per second to `Particle.connect`.  Leave this in place for now, we need to investigate whether the rate limiting is still a concern.

closed 2025-02-05 by ntlhui