UCSD-E4E/smartfin-fw3 #173: CLI Backspace for MacOS

ezh247467, 2025-02-11:
**Describe the bug**
For MacOS users, the "backspace" button is mapped to DEL, which is 127 in the ASCII table.

**To Reproduce**
Steps to reproduce the behavior:
1. Build the ```smartfin_pc``` build
2. ```Run build/pc_hal/smartfin_pc```
3. Enter ```#CLI``` without a trailing newline
4. Press backspace/delete
5. See error

**Expected behavior**
Backspace in the command line

**Screenshots**

**Hardware Configuration(please complete the following information):**
 - Hardware Version [PCB revision]
 - Potted [Y/n]
 - Firmware Version

ntlhui, 2025-02-11:
Please add the firmware version this was discovered in (either git revision or semantic version)

Note that this was seen on the Docker build.

Please add information about which MacOS version and hardware you are running on.

ezh247467, 2025-02-11:
> Please add the firmware version this was discovered in (either git revision or semantic version)
> 
> Note that this was seen on the Docker build.
> 
> Please add information about which MacOS version and hardware you are running on.

I've edited the issue to include the information you mentioned. Do I need to add more details about the MacOS and hardware?

ntlhui, 2025-02-11:
That information is fine.

ntlhui, 2025-02-12:
Can anyone in @UCSD-E4E/smartfin replicate this behavior on Apple silicon mac, Linux, Windows, or firmware?

ntlhui, 2025-02-14:
@ezh247467 Which keyboard and keyboard layout do you have on your system?

ezh247467, 2025-02-14:
@ntlhui I use the MacBook Pro 2019 keyboard with the U.S. input source keyboard layout.

ntlhui, 2025-03-02:
On Windows 10, Docker v4.38.0, FW 3.7.0
```
Smartfin FW v3.7.0
FW Build: Mar  2 2025 01:36:06
Device OS:
     1: display Menu       
     2: disconnect particle
     3: connect particle
     4: show flog errors
     5: test printf
     6: debug menu
     7: hexdump
     8: gps
     9: sleep
    10: Self Identify
    11: check charge ports
    12: MFG Test
    13: upload
    14: Recorder Test Menu
    15: Session Test Menu
   100: Set State
   101: Display System State
   102: Display NVRAM
   200: Sleep - Set Sleep Behavior
   201: Sleep - Get Sleep Behavior
   300: Display Reset Reason

>^?^?^?^?^?asdf^?^?^?^?^?
```
@ezh247467 is this the behavior you get?  I get a `^?` char every backspace I hit.

ezh247467, 2025-03-02:
@ntlhui Yes, I get a `^?` char every backspace I hit as well.

ntlhui, 2025-03-02:
This is not a MacOS issue, this is likely a ncurses issue.

ezh247467, 2025-03-02:
Okay, the resolution for my device was to include `case 127:` since 127 is DEL in ASCII. Does that work for you?

ntlhui, 2025-03-02:
Create a pull request

PR #179 (2025-03-02): Include DEL (127 in ASCII) input to be recognized as backspace
Fixes #173 by recognizing DEL input as backspace.
commit eebe430f (2025-03-02): Include DEL (127 in ASCII) input to be recognized as backspace (#179)

Fixes #173 by recognizing DEL input as backspace.
closed 2025-03-02 by ezh247467