UCSD-E4E/smartfin-fw3 #183: ubloxGPS Malloc

ezh247467, 2025-03-31:
In `./lib/gps-ublox/src/ubloxGPS.cpp`, `malloc` is used. Flagged for @ntlhui to take a look.

ntlhui, 2025-03-31:
Combine resolution with #120

ntlhui, 2025-05-15:
Assuming `free` is implemented properly, this is probably safe.  `malloc` and their corresponding `free` is only invoked when `UBX-MON-VER` is polled from the uBlox receiver, so functionally it shouldn't be called.  In the event it is called, the `mallocs` together will allocate at most 70 bytes (not including padding), which are held by a private member struct.  No code within the struct re-uses those addresses.

ntlhui, 2025-05-15:
Reference https://content.u-blox.com/sites/default/files/products/documents/u-blox8-M8_ReceiverDescrProtSpec_UBX-13003221.pdf?utm_content=UBX-13003221 32.16.13.2

closed 2025-05-15 by ntlhui