UCSD-E4E/smartfin-fw3 #225: Consider upgrading to DeviceOS 6.2.0 or later

ntlhui, 2025-05-07:
https://docs.particle.io/reference/device-os/api/publish/limits-publish/#limits-publish

This supports binary data

We might be able to avoid the base85 overhead!

ntlhui, 2025-05-07:
@SUPScientist @rck289 thoughts on this?

We would be able to upload 1 kB binary blobs.  We currently can only upload 768 bytes at a time.  This would be a 33% increase in upload efficiency in terms of cost.  This would break compatibility with existing data interfaces.

ntlhui, 2025-05-07:
There is also an opportunity to switch to DeviceOS >6.3.0, in which we can upload 16 kB blobs, however, we would need to use a different API, so would double break compatibility.

This is also buying more into the cellular infrastructure, as opposed to pushing more towards the BLE interface we had been thinking of a while back.

SUPScientist, 2025-05-07:
Nothing against upgrading to DeviceOS 6.2.0 (or higher) in general. Switching to binary and fallout effects sound like too much extra work to justify prioritizing this now. I suggest keeping a switch to more efficient binary on the backburner (backlog?) but not undertaking now.