UCSD-E4E/smartfin-fw3 #230: 100 Hz IMU

ntlhui, 2025-05-09:
@avikg511 requesting 100 Hz DMP output, ideally 400 Hz Accel/Gyro and 100 Hz Mag

This will publish as IMUx3 (ENS09)

ntlhui, 2025-05-14:
Currently working on implementing a separate acquisition loop for the IMU.

avikg511, 2025-05-16:
Attached below is first draft of Data Request:

[SmartfinDataRequest.pdf](https://github.com/user-attachments/files/20259705/SmartfinDataRequest.pdf)

Under the accelerometer registers, I explained an overall confusion with configuring the Low Pass Filter cutoffs, any thoughts?

ntlhui, 2025-05-17:
You haven't defined \gamma!

ntlhui, 2025-05-17:
We can look at the system behavior under either lpf condition. I think this is implemented in the sensor firmware. Odr should set the sampling rate, lpf applied after

ntlhui, 2025-05-17:
We are not able to access the raw ADC from the sensor

avikg511, 2025-05-20:
What else do you want finalized before data collection though?

avikg511, 2025-06-04:
+/- 4g should be fine! Just keeping this open for now as a reminder to save the updated pdf. Haven't had the time to sit down and make the changes yet but they'd be good for future documentation